e-tinkers/esp32-adc-calibrate #1: 'adcStart' was not declared in this scope

chinswain, 2020-07-16:
Hi,
I get the below errors when running the code in Arduino 1.8.13 with the latest arduino-esp32 build:

 ```
error: 'adcStart' was not declared in this scope
error: 'adcEnd' was not declared in this scope
```

e-tinkers, 2020-07-17:
What is GPIO pin value you pass in to the `adcStart()` function? Pin 35 that I used on a generic [ESP32 Dev Baord](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap), is equivalent to A7 in Arduino. If you pass in an invalid pin (e.g. A1 or A2), you will get the error as those are not defined in ESP32 as valid ADC inputs.

chinswain, 2020-07-17:
I've tried a few boards including a generic one, I've checked the schematics on a few to confirm the pins are correct. The error occurs when I compile against "ESP Dev Module" so the pin mapping should be 25/35 as your example.

[image]

I've posted a message in the arduino-esp32 chat, I'll update here if there's any news.

chinswain, 2020-07-17:
It seems they were removed, would you be able to update the code to reflect the change (to analogRead())?

https://github.com/espressif/arduino-esp32/commit/d8b209846131119ad6a84af5987e8625f37900df

e-tinkers, 2020-07-18:
Interesting. I've modified the code to use analogRead() instead. Do a `git pull` or download the new version.

closed 2020-07-21 by e-tinkers