e-tinkers/esp32-adc-calibrate #2: Doubt about accuracy of the DAC itself

A-R-S-D, 2021-02-16:
Hi, thanks for this great effort.

According to my understandings in order to calibrate a device we should use a more accurate and precise device as reference, but here how would you know that DAC of ESP32 is accurate and precise enough to do so? What if the DAC do suffer from the same non Linearity issues as ADC?  I mean then have you tested the calibrated ADC with oscilloscope and precise voltage generators or something to see whether the results improved or worsened?

Sorry that i am not trying it myself because currently i haven't required tools to do so.

europer, 2021-03-13:
I use the DAC my self (in my case pin 25) and can see that the DAC values from 0-255 are not constant in many ways 1. From 0 - 255 = 0 - 3.3v is in it self not linear. 2. What is an observation is everything i Power up the ESP32, lets say with the value 127 I get different voltage and if I change power supply anyway...it's so strange the DAC so I agree your using the DAC for reference but it is in it self inconstant and not linear. I was thinking off using a LED (or LEDs ) for reference voltage and have measured them to be very constant with almost no deviations but don't know yet how to implement this. Since I use the DAC I would use the ADC to measure the DAC to adjust if the DAC is giving false voltage but need then a ADC what is adjusted to the LED reference voltage and all this get complicated and lot of pins

A-R-S-D, 2021-03-14:
How using LED can make measurements more precise. Neither using DAC nor using DAC with LED helps. In fact if DAC is erroneous and its error is the same as ADC or even worse that ADC (which i think IT IS) using this method just make it worse and complicated.

e-tinkers, 2021-03-14:
The purpose of this project was to make it reasonably accept and good enough that I can use the ESP32 in my project my project (measure temperature or something do not need supper accuracy). If I would need the precision and accuracy, I won't use ESP12 for my ADC, an external MCP3221 12-bit I2C chip would be my approach or replace ESP32 with STM32, if I need WiFi, I'd just added an ESP01 as the WiFi shield as I described in my [blog](https://www.e-tinkers.com/2020/04/a-better-way-to-use-esp-01-as-wifi-shield/).

In fact, now I think what I did in this repo might be just "over-kill" for some of the projects. For example, if I'm only interested on temperature between 10 to 30 degree Celsius, I would use a multimeter and some known resistors and calculate the temperatures at each resistor value, and with a much smaller look up table...

```
float read_thermistor() {
  int i, cntx;   
  float temp, m, b;  // variables for line equation 
  // look up table to convert counts to temperature  
  int cnt[5] = {1163, 1569, 1657, 1734, 2021}; 
  float temp_point[5] = {8.5, 18.0,  21.0, 23, 30.5};  

  cntx = analogRead(34);

  //  determine which two point to used based on cntx
  if (cntx < cnt[1]  ) { i = 0; }    
  if(cntx <= cnt[1] ) i = 1; 
  if(cntx <= cnt[2] ) i = 2;  
  if(cntx <= cnt[3] ) i = 3;

  // Calculate slope and constant for line  :       temp = mx + b
  m = (temp_point[i + 1] - temp_point[i] )  /  ( cnt[i+1] - cnt[i] );
  b = temp_point[i] - ( m * cnt[i]  );
  temp = m * cntx + b;
  return(temp);
}
```

e-tinkers, 2021-03-14:
> I was thinking off using a LED (or LEDs ) for reference voltage 

First of all, ESP32 does not support any external voltage reference like Arduino does. Secondly, even it does, you need a high precision voltage reference such as [MCP1501](https://www.microchip.com/wwwproducts/en/MCP1501) provide the inaccuracy was caused by the instability of reference voltage.

zeyadnabhan, 2026-10-09:
Good question, and it still applies: a DAC-referenced table makes the DAC→ADC pair linear, not the ADC alone, so DAC error ends up in the table. To at least quantify the result, I added a verification pass to a rewrite of this generator. It re-sweeps the DAC and reports the error before and after correction (on my board: raw rms 212.8 → 4.1 LSB). For absolute accuracy you still need a multimeter or an external reference. https://github.com/zeyadnabhan/esp32-adc-lut