e-tinkers/esp32-adc-calibrate #4: Use a more precise signal reference

maxmeli123, 2023-03-02:
Because not to use LEDC or I2S PDM to generate a precise signal reference instead of low (8 bit) resolution internal DAC  ?

Frequency   |     Bit depth   |   Available steps
Max 1220Hz          16                     65536
Max 2441Hz          15                     32768
Max 4882Hz          14                     16384
Max 9765Hz          13                       8192
Max 19531Hz        12                       4096

You can eg generate a frequency max 1220Hz and have a resolution of 16 bits changing the dutu cycle from 0 to 65535.

LEDC is linear and use ESP 16 bit internal timers, it is precise and you can setup it as you like by setting resolution, frequency and finally the desired duty cycle.

I used LEDC in 1bit mode to generate frequencies up 40Mhz to use as master clock signal instead a quartz, to sample audio over I2S  so it can generate a precise signal reference on almost any pin, check documentation to know pins that support it.

To improve your project I suggest you these things:
- add a small poliestere (or ceramic) capacitor from ADC input pin to GND (like a 100nF), this depends on PWM frquency you use, the higher the frequency  the smaller the capacitor should be. To be precise the better is to generate a 50% duty cycle, use an adjust capacitor, and check it to an oscilloscope, adjust it to have just a line on oscilloscope, so until you measure around 0 volts, at the end some small noise will remain, this not a problem, eventually you can put some other smaller ceramic capacitors in parallel. At the end by check the signal on serial plotter you should see an exact line, no noise at all, you now use 4096 steps not 256 steps, and with capacitor it should be the difference.
- for your cpp file try to import <driver\ledc>, initialize it... and then in the sampling cycle you can just change the duty cycle every step. If you eg use 16 bits at 1000 Hz, just divide 65536 by 4096 that are max steps for ADC, doing this you find the step to increase a duty cycle in a loop of 4096 steps (that is 16 units).
-  in my opinion to have better accurate measurements you should add a significant delay after you acquire every of 4096 values, like at least 20-50 milliseconds, this help to be more precise depending if you use float or integer math for a lookup table. Optionally sample 10 or more times and then average it help to remove noise. Optionally apply a software LPF filter will remove completely tha ADC noise. I use it and works very well. If you want some code for this I can send you, the code is simple, just a few of lines. I use it on many platforms even when I develop for desktop machines.

Another way slould be to use I2S and setup it in the I2S config as PDM out. This even is used for audio, so it is good and can be used as linear precise voltage reference, more more better than internal 8 bit ADC, it is loke default Raspberry 3.5  mini audio jack, not so good like external audio DAC but can play 16 bit audio.

I'm not sure if this help you, I hope yes.
Many thanks for your work.
Max.

maxmeli123, 2023-03-02:
Check here at the voice 'Minimize noise', as you can see are exact my advices, use a 100nF to GND and sample more samples, then average the result, possibly add a small delay between any sample acquire will increase accuracy:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc_calibration.html#minimize-noise

Sorry if I used another message, and sorry if a link is just a text, but my actually pc browser do not allow it, only my phone can do it.

techstudio-design, 2023-03-03:
Thanks for the comment. The idea of using LEDC to generate as a liner voltage reference is certainly an interesting one and I know nothing about it until you mentioned it, I will certainly give it a try when I have time. 

To be honest, I haven't touch the ESP32 ADC since 3-year ago when I was doing these testings. Nowadays, I either uses an external 16-bit ADC chip with I2C interface with a separate power rail for serious work (for clients who need precision ADC), or for hobby and quick-and-dirty project, using a much simple 5 sampling points at different voltages, and create an array like `adcValues[5]{{1163, 1569, 1657, 1734, 2021}` for each voltage measurement. It is much simpler than going through the elaborated way that I was doing several years ago.

maxmeli123, 2023-03-23:
Hi, thanks for reply and sorry for delay on my reply,

I managed it a bit to work with a LEDC as precise reference managing 4096 values.
Here a first attemp to do it, note that I used some digital LOW PASS FILTERS to search to remove noise but may these are not necessary.
If you see the code I replaced the internal DAC (8 bit 256 values with LEDC) and setup it to the same resolution to get 4096 values.

To use this I've put an RC to output pin, like a resistor in serie and a small capacitor in parallel.
Anyway it is an high frequency PWM an need to be filtered.
Because I do not know exact values I've put a trimmer and an adjustable capacitor.
I've tried to get a more clean signal as possible but just using a serial monitor to get results is not the right way to do this, need an oscilloscope (and I don't have it).

Note that before this and after I tried your code, I've wrote an ESP32 library based on it, I called it
ESP32LinearADC, it works like your code and it is very simple to use, something like that;

#include <ESP32LinearADC>

ESP32LinearADC ADC;

Serial.print ("Initializing ADC ...  ");
if (! ADC.begin()) {   // Begin ADC
      Serial.println ("Error while initialize ADC");
      while (1) delay (1);
}

// JUncomment this line once to calibrate ADC and show ADC LUT on serial monitor, then 
// copy it once to User_Setup.h library header. After this any time you use ADC will return
// linear values.

uint16_t value = ADC.read(35);  // This return our linear raw value (0 to 4096, note that is has a small gap under 50 and highter than 4030)  no need to do anything else

float voltage = ADC.readVoltage(35,  3.3);  / // Pin, vRef

This worked well, as your code, but now I want improve it by using LEDC with true 4096 values.
If you want to do a try here some code you can adapt. I do not removed comments, so maybe can be useful for you.... my apologies if I cannot put the code in a coe tag, but I'm here with an old PC and my browser refuse to do it.

I hope it is useful for you, at least it will show how use LEDC to get it working.
If you try and have success, please reply here your successes.

Many thanks

//#include <Arduino.h>
//#include <driver/dac.h>  // REPLACED WITH LEDC WHERE WE SET 12 BIT RESOLUTION

// Based on original work from Helmut Weber (https://github.com/MacLeod-D/ESP32-ADC)
// that he described at https://esp32.com/viewtopic.php?f=19&t=2881&start=30#p47663
// Modified with bug-fixed by Henry Cheung
//
// Build a ESP32 ADC Lookup table to correct ESP32 ADC linearity issue
// Run this sketch to build your own LUT for each of your ESP32, copy and paste the
// generated LUT to your sketch for using it, see example sketch on how to use it
//
// Version 2.0 - switch to use analogRead() instead of esp-idf function adcStart()
// Version 1.0 - original adoptation and bug fix based on Helmut Weber code

//#define GRAPH      // uncomment this for print on Serial Plotter
//#define FLOAT_LUT     // uncomment this if you need float LUT
#define ADC_PIN 33  //35    // GPIO 35 = A7, uses any valid Ax pin as you wish

#define LEDC_CHANNEL        0   // Use first channel of 16 channels (started from zero)
#define LEDC_RES           12   // Use 12 bit precision for LEDC timer
#define LEDC_FREQ        5000   // Use 5000 Hz as a LEDC base frequency
#define LED_PIN          25   // Fade LED PIN (replace with LED_BUILTIN constant for built-in LED to test)

#define SAMPLES ((int)(pow(2, LEDC_RES)))
#define AVERAGE_SAMPLES    40

uint16_t TIME_BEFORE_SAMPLE;  // In ms (time required to stabilize before samples are taken)
#define SAMPLING_TIME       100  // In us  100

#define MAX_DUTY_CYCLE ((int)(SAMPLES-1))
[... 237 lines cut ...]
   //
   //   Serial.println();

#ifdef FLOAT_LUT
   Serial.println ("const float ADC_LUT[4096] = { 0,");
   for (int i = 0; i < MAX_DUTY_CYCLE; i++) {  ///  ERA DA 0 A 4095
      Serial.print (Results[i], 4); Serial.print (",");
      if ((i % 15) == 0) Serial.println();
   }
   Serial.println (Results[MAX_DUTY_CYCLE]);
   Serial.println ("};");
#else
   Serial.println ("const int ADC_LUT[4096] = { 0,");
   for (int i = 0; i <= MAX_DUTY_CYCLE; i++) { ///  ERA DA 0 A 4095
      Serial.print ((int)Results[i]); Serial.print (",");
      if ((i % 15) == 0) Serial.println();
   }
   Serial.println ((int)Results[MAX_DUTY_CYCLE]);
   Serial.println ("};");
#endif

   while (1);

   //#endif
}

e-tinkers, 2023-03-24:
Hi, appreciated your effort. Can you clean up your code by removing those unused code, with correct indentation? Code block has nothing to do with how old/new is your PC, github uses markdown language, to put a block of code in a code block, you add three backticks ``` at the beginning and ending of the code. 

Since this is not an issue related to my original git repository but a suggestion or an alternative way of achieving the same result. I'm closing this issue for now.

maxmeli123, 2023-05-07:
Hi, 
there are some develops ? 
Did you tried my code ?

Thanks
Max